davidcaron/pclpy #8: Is it possible to convert las file to pcd file using pclpy?  :)

niranjanreddy891, 2018-05-15:
I am using win-7 64 bit with python-3.6 installed

davidcaron, 2018-05-15:
That should be possible of course. I'll try to add a test and post it here when I get a chance.

davidcaron, 2018-05-15:
I added a simple test showing how it can be done. Here is the idea:

``` python
pc = pclpy.io.las.read("input_cloud.las", "PointXYZ")

# the rest is very close to the PCL api
writer = pcl.io.PCDWriter()
writer.writeBinary("cloud.pcd", pc)
```

closed 2018-05-15 by davidcaron

niranjanreddy891, 2018-05-15:
How to try this. I mean the sample code which you gave. :)

niranjanreddy891, 2018-05-15:
I modified the code little bit. can you check it once. :)

davidcaron, 2018-05-15:
try this:

```python
import pclpy
from pclpy import pcl

pc = pclpy.io.las.read("E:/Lidar-practice/new-files/las14_type7_ALS_RIEGL_Q680i.las", "PointXYZ")
writer = pcl.io.PCDWriter()
writer.writeBinary("E:/Lidar-practice/new-files/qwerty.pcd", pc)
```

or the same code, but with a function call at the bottom:

```python
import os
import time

import pytest
import numpy as np
import laspy

from pclpy import pcl
import pclpy

def test_las_rgb():
    pc = pclpy.io.las.read("E:/Lidar-practice/new-files/las14_type7_ALS_RIEGL_Q680i.las", "PointXYZ")
    assert pc.size() == 5025
    assert np.any(pc.r)

    writer = pcl.io.PCDWriter()

    writer.writeBinary("E:/Lidar-practice/new-files/qwerty.pcd", pc)

test_las_rgb()
```

niranjanreddy891, 2018-05-15:
The code is working fine. But the output pcd file is empty.

davidcaron, 2018-05-15:
Try changing `"PointXYZ"` to `"PointXYZRGBA"` when reading your las file

niranjanreddy891, 2018-05-15:
Thanks @davidcaron

HGSipahi, 2021-12-03:
Hello, first of all thanks for the application. I am having some problems while using it. When I run the python file to convert the las file to pcd with the code shared above, I get the error **"Exception has occurred: ModuleNotFoundError"**-"**No module named 'pclpy.pcl'"**