davidcaron/pclpy #12: Segmentation code not working

niranjanreddy891, 2018-05-21:
[check this link in the repository](https://github.com/davidcaron/pclpy/blob/master/pclpy/tests/test_visualization.py)

davidcaron, 2018-05-23:
This code was actually me playing with the min cut segmentation. It was commented out, and I just removed it.

I suggest reading about the functions you're trying to use first... You're copying and pasting code and it seems you didn't think about what's actually going on. (For example you kept some point coordinates which I'm sure are outside of your cloud's bounding box).

closed 2018-05-23 by davidcaron

niranjanreddy891, 2018-05-24:
Still the segmentation code is not working. Do I need to modify anything in the code. :)
```
import math
import os
import numpy as np
import pytest

import pclpy
from pclpy import pcl

def test_data(*args):
    return os.path.join("test_data", *args)

def make_pt(x, y, z):
    pt = pcl.point_types.PointXYZRGBA()
    pt.x = x
    pt.y = y
    pt.z = z
    return pt

def test_region_growing():
    pc = pclpy.io.read(test_data("E:/Lidar-practice/new-files/las14_type7_ALS_RIEGL_Q680i.las"), "PointXYZRGBA")
    rg = pcl.segmentation.RegionGrowing.PointXYZRGBA_Normal()
    rg.setInputCloud(pc)
    normals_estimation = pcl.features.NormalEstimationOMP.PointXYZRGBA_Normal()
    normals_estimation.setInputCloud(pc)
    normals = pcl.PointCloud.Normal()
    normals_estimation.setRadiusSearch(0.35)
    normals_estimation.compute(normals)
    rg.setInputNormals(normals)

    rg.setMaxClusterSize(1000000)
    rg.setMinClusterSize(10)
    rg.setNumberOfNeighbours(15)
    rg.setSmoothnessThreshold(5 / 180 * math.pi)
    rg.setCurvatureThreshold(5)
    rg.setResidualThreshold(1)
    clusters = pcl.vectors.PointIndices()
    rg.extract(clusters)
    assert max([len(c.indices) for c in clusters]) == 2449  # ground
```

davidcaron, 2018-05-24:
Are you running this code *exactly* like you pasted it? Because the function won't be executed.

Also, you're keeping the exact same parameters from the tests again. Did you check if they make sense with your cloud?

niranjanreddy891, 2018-05-24:
But why the function can't be executed?

davidcaron, 2018-05-24:
Add `test_region_growing()` at the end of the script. This is python 101. I'm sorry but I can't find the time to help on everything. You will have to show that you tried something on your own.

niranjanreddy891, 2018-06-12:
@davidcaron, any update on segmentation?