davidcaron/pclpy #13: Unable to fit the output to the window properly?

niranjanreddy891, 2018-05-23:
Hi, When I am converting las file to pcd file, then visualizing converted pcd file, I am unable to fit the output to the window properly. I tried using `R` in the keyboard. For zooming I used mouse and also `F`.
A small dot is being displayed in the output. Please go through the output which I have attached.
[image]
window. But it couldn't help me. Can you please solve the issue?

This is the file. It has both las file and the converted PCD file. 

This is the code which I am trying to use.

```
import os
import time

import pytest
import numpy as np
import laspy

from pclpy import pcl
import pclpy

def test_las_rgb():
    pc = pclpy.io.las.read("E:/Lidar-practice/new-files/las14_type7_ALS_RIEGL_Q680i.las", "PointXYZ")
    assert pc.size() == 5025
    assert np.any(pc.r)

    writer = pcl.io.PCDWriter()

    writer.writeBinary("E:/Lidar-practice/new-files/qwerty.pcd", pc)
```

[las14_type7_ALS_RIEGL_Q680i.zip](https://github.com/davidcaron/pclpy/files/2031560/las14_type7_ALS_RIEGL_Q680i.zip)

davidcaron, 2018-05-23:
and where is the code you're using to view the cloud? are you using `viewer.resetCamera()`?

niranjanreddy891, 2018-05-24:
This is the code which I am using for visualizing pcd file. 
```
from __future__ import print_function
import numpy as np
import pcl
import pcl.pcl_visualization

cloud = pcl.load_XYZRGB('E:/Lidar-practice/new-files/testing/las14_type7_ALS_RIEGL_Q680i.pcd')
visual = pcl.pcl_visualization.CloudViewing()

visual.ShowColorCloud(cloud, b'cloud')

visual.resetCamera()
flag = True
while flag:
    flag != visual.WasStopped()
end

```