davidcaron/pclpy #14: How to handle errors if the file does not have red color

niranjanreddy891, 2018-05-30:
Lets say I have a las file, it has blue and green colors but no red color. How to display the file with the colors it has. i.e; if the file doesn't have red color, instead of throwing `Dimension red not found` error, it has to display the colors as it is in the file. How to do it

As a reference I have this file
[las14_type6_ALS_RIEGL_VQ580.zip](https://github.com/davidcaron/pclpy/files/2053086/las14_type6_ALS_RIEGL_VQ580.zip)

niranjanreddy891, 2018-06-06:
@davidcaron, can you help me in this?

davidcaron, 2018-06-11:
Well that's an unusual case... 

If your file has only blue and green, and the red field is absent, it doesn't comply with the las standard... 

But that's still possible. So in that case, you would have to create a `red` field with zeroes in it. Then the file will display just fine.

`Dimension red not found` is a laspy error because red is the first color dimension it's looking for.

niranjanreddy891, 2018-06-11:
Thanks for the reply@davidcaron, the file which I uploaded above doesn't have red color. So, how to know a file follows `las standard` ?

niranjanreddy891, 2018-06-11:
Also, what if the file has only orange color. See this file
[305500_5750500.zip](https://github.com/davidcaron/pclpy/files/2090795/305500_5750500.zip)
How to handle this type of files?
I am curious to know about it. :)

davidcaron, 2018-06-11:
> the file which I uploaded above doesn't have red color

The file doesn't have any color. If you try to read its colors, it's normal that you get an error.

> Also, what if the file has only orange color. See this file

The possible colors are red, green and blue. Also, this other file doesn't have any color either.

closed 2018-06-11 by davidcaron

niranjanreddy891, 2018-06-12:
@davidcaron, how to know if a file follows las standard or not?