davidcaron/pclpy #15: getRemovedIndices returns empty Int vector

ruoyu0088, 2018-07-20:
I want to call `getRemovedIndices()` to get the index of removed points:

```
from pclpy import pcl
import numpy as np
arr = np.random.randn(30, 3).astype(np.float32)
cloud = pcl.PointCloud.PointXYZ.from_array(arr)
cloud_filtered = pcl.PointCloud.PointXYZ()
pas = pcl.filters.PassThrough.PointXYZ()
pas.setInputCloud(cloud)
pas.setFilterFieldName("z")
pas.setFilterLimits(0, 1)
pas.filter(cloud_filtered)
print(pas.getRemovedIndices())
```

but the result is `Int[]`.

davidcaron, 2018-07-20:
It seems the problem is with the `filter` method when given a point cloud as parameter.

If you give an Int vector to the filter method, it works:

``` python
# this is the same as before
from pclpy import pcl
import numpy as np
arr = np.random.randn(30, 3).astype(np.float32)
cloud = pcl.PointCloud.PointXYZ.from_array(arr)
cloud_filtered = pcl.PointCloud.PointXYZ()
pas = pcl.filters.PassThrough.PointXYZ()
pas.setInputCloud(cloud)
pas.setFilterFieldName("z")
pas.setFilterLimits(0, 1)

# passing a pcl.vectors.Int() to filter() is a workaround...
indices = pcl.vectors.Int()
pas.filter(indices)
print(indices)
```

But the filter function should work with a point cloud, so this is a bug.

davidcaron, 2018-08-06:
Ok, I figured it out. It's a simple PCL flag that wasn't set: `extract_removed_indices=True`

``` python
from pclpy import pcl
import numpy as np
arr = np.random.randn(30, 3).astype(np.float32)
cloud = pcl.PointCloud.PointXYZ.from_array(arr)
cloud_filtered = pcl.PointCloud.PointXYZ()
pas = pcl.filters.PassThrough.PointXYZ(extract_removed_indices=True)  # only this line changed
pas.setInputCloud(cloud)
pas.setFilterFieldName("z")
pas.setFilterLimits(0, 1)
pas.filter(cloud_filtered)
print(pas.getRemovedIndices())
```

Also, you could do the same thing using numpy logic, but keep in mind this should be slightly slower:

``` python
from pclpy import pcl
import numpy as np
arr = np.random.randn(30, 3).astype(np.float32)
cloud = pcl.PointCloud.PointXYZ(arr)
xyz = cloud.xyz

indices = np.where(np.logical_and(xyz[:, 2] > 0, xyz[:, 2] < 1))[0]
print(indices)

# and create a new cloud if necessary
cloud_filtered = pcl.PointCloud.PointXYZ(xyz[indices])
```

closed 2018-08-06 by davidcaron