davidcaron/pclpy #16: Getting points from PointXYZ obtained from numpy array

ostroznie, 2018-07-22:
I am working with point cloud data from RealSense sensor, which I store in numpy array. I am reading it through `cloud.from_array(data)`, where cloud is PointXYZ object. I want to use filtering and feature extraction functionalities from pclpy on this data, but it seems that PointXYZ doesn't return any coordinates. Printing `cloud.xyz` returns []. I also tried generating random numpy array, but it yields the same results. Ultimately I would like to process this cloud and then return modified cloud back to numpy array. Am I missing something?

davidcaron, 2018-07-22:
The first thing that comes to mind is that the array given to `from_array` has to be contiguous, with one point on each row.

Try:
```python 
data = your_array.reshape(-1, 3)
xyz = np.ascontiguousarray(data, dtype=np.float32)
pc = pcl.PointCloud.PointXYZ.from_array(xyz)
```

I think pclpy should raise an exception if that's not the case.

davidcaron, 2018-07-22:
Oh and the from_array method is a staticmethod. So make sure to call it like this:
`pcl.PointCloud.PointXYZ.from_array(xyz)`

Or this does the same:
`pcl.PointCloud.PointXYZ(xyz)`

closed 2018-08-06 by davidcaron

szx0112, 2020-05-15:
Hi @davidcaron, I am trying to assign both points and colors to a point cloud array. How could we do it?

I tried:  
xyz as points nx3 numpy array in float32, rgb as nx3 numpy array in uint8
pcd = pcl.PointCloud.PointXYZRGB([xyz,rgb])
or
pcd = pcl.PointCloud.PointXYZRGB(xyz,rgb)

But none of the above success. I could not find any document related to color assignment in the community yet.