davidcaron/pclpy #18: Is there any documentation or tutorials about this library?

yangj257, 2018-07-26:
Except for the very simple example, I didn't find any documentation or tutorials about this library. Can you edit one just like laspy, which allows beginners to get started quickly and make your project get more attention. You can start with simple modules, like I/O,kd-tree,octree,visualization,etc. Thank you very much!

davidcaron, 2018-07-26:
Yes, it's definitely needed. My focus was on getting it to work first. I'll write a first draft when I get a chance.

yangj257, 2018-07-27:
I hope that the first draft can be completed as soon as possible, I can't wait for it!
You have contributed a lot to python and pcl!

pournami123, 2019-05-24:
@davidcaron Is there any documentation available?  please share

ellie-b, 2020-01-21:
I would also like to see documentation to get started using this @davidcaron

OloOcki, 2020-01-21:
up! :)

ellie-b, 2020-01-21:
@OloOcki Where is the additional documentation?

OloOcki, 2020-01-25:
There is no additional documentation. That is why I wrote "up!" ;)

soulslicer, 2020-01-31:
Yet another library that lacks proper documentation. Just like strawberry labs garbage pcl port, pptk, and open3d. all lack even being able to do a simple thing like visualzing a color pointcloud in real time in python.

I ended up writing by own pybind11 bindings to the PCL C++ API and am using that now.

glassTransition, 2021-03-13:
> I ended up writing by own pybind11 bindings to the PCL C++ API and am using that now.

I really don't understand why these develops both to "provide" binding if they have no intention of making the libraries usable.

I wish I had the skills to write my own bindings, but I'm trying to use this library because I only know Python. I don't have the time to learn C++