davidcaron/pclpy #19: FPFH descriptor

pradAI, 2018-08-23:
Hi,
Thank you for providing pcl library in Python.

I am planing to use extract features using FPFH algorithm, can you please help me regarding this. I am 
 trying to load x,y,z,intensity but getting following error.

import pclpy
from pclpy import pcl
​
point_cloud = pclpy.read("ground.las", "PointXYZI")
mls = pcl.surface.MovingLeastSquaresOMP.PointXYZI_PointNormal()
​
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-24-e7702617a15b> in <module>()
      3 
      4 point_cloud = pclpy.read("ground.las", "PointXYZI")
----> 5 mls = pcl.surface.MovingLeastSquaresOMP.PointXYZI_PointNormal()

**AttributeError: module 'pclpy.pcl.surface.MovingLeastSquaresOMP' has no attribute 'PointXYZI_PointNormal'**

How can I load X,Y,Z and Intensity?

Thank you

davidcaron, 2018-08-23:
Only the classes compiled in the Windows PCL release are available. And unfortunately the `MovingLeastSquaresOMP<PointXYZI, PointNormal>` isn't compiled so it isn't available.

Using an IDE that has autocompletion or ipython or `dir(pclpy.pcl.surface.MovingLeastSquaresOMP)`, you can see which types are compiled:

```
['PointNormal_PointNormal',
 'PointNormal_PointXYZ',
 'PointNormal_PointXYZRGB',
 'PointNormal_PointXYZRGBA',
 'PointNormal_PointXYZRGBNormal',
 'PointXYZRGBA_PointNormal',
 'PointXYZRGBA_PointXYZ',
 'PointXYZRGBA_PointXYZRGB',
 'PointXYZRGBA_PointXYZRGBA',
 'PointXYZRGBA_PointXYZRGBNormal',
 'PointXYZRGBNormal_PointNormal',
 'PointXYZRGBNormal_PointXYZ',
 'PointXYZRGBNormal_PointXYZRGB',
 'PointXYZRGBNormal_PointXYZRGBA',
 'PointXYZRGBNormal_PointXYZRGBNormal',
 'PointXYZRGB_PointNormal',
 'PointXYZRGB_PointXYZ',
 'PointXYZRGB_PointXYZRGB',
 'PointXYZRGB_PointXYZRGBA',
 'PointXYZRGB_PointXYZRGBNormal',
 'PointXYZ_PointNormal',
 'PointXYZ_PointXYZ',
 'PointXYZ_PointXYZRGB',
 'PointXYZ_PointXYZRGBA',
 'PointXYZ_PointXYZRGBNormal']
```

A workaround would be to put the intensity in the RGB field of a new point cloud maybe?

Something like:

```python
import pclpy
from pclpy import pcl
​
point_cloud = pclpy.read("ground.las", "PointXYZI")

xyz = point_cloud.xyz
uint8_intensity = np.array(point_cloud.intensity * 256, dtype="u1")
rgb = np.array([uint8_intensity, uint8_intensity, uint8_intensity], dtype="u1").T
point_cloud_2 = pcl.PointCloud.PointXYZRGBA(xyz, rgb)

mls = pcl.surface.MovingLeastSquaresOMP.PointXYZRGBA_PointNormal()
mls.setInputCloud(point_cloud_2)
# other operations...
```

It's untested, but I hope you get the idea.

pradAI, 2018-08-24:
Thank you for your help. it is working, How to extract FPFH descriptors using pclpy

davidcaron, 2018-08-24:
Here is a sample that should compute FPFH descriptors (however, the octree_voxel_downsample function should only work in v0.10.0 which I will release in the next hours):

```python
    pc = pclpy.read(test_data("street.las"), "PointXYZ")

    pc = pclpy.octree_voxel_downsample(pc, 0.25)
    normals = pc.compute_normals(radius=0.5, num_threads=8)

    fpfh = pcl.features.FPFHEstimation.PointXYZ_Normal_FPFHSignature33()
    fpfh.setInputCloud(pc)
    fpfh.setInputNormals(normals)
    fpfh.setRadiusSearch(0.5)

    fpfhs = pcl.PointCloud.FPFHSignature33()
    fpfh.compute(fpfhs)

    print(fpfhs.histogram)
```

However, I'm not familiar with descriptors, so I can't confirm that the results are actually what's expected :)

It seems to work though, as far as I know.

pradAI, 2018-08-27:
Great David, Thanks again for your time and help.