davidcaron/pclpy #20: GreedyProjectionTriangulation

lsymuyu, 2018-08-24:
Thanks for your nice work. Otherwise I wouldn't have chance to try PCL. (No experience of using C).
I am trying to triangulate a point cloud. As the first trial, I am trying to use GreedyProjectionTriangulation method. Here is the code:
## Triangulation
Points = np.concatenate((PCL_filter3.xyz,
                         PCL_filter3.intensity.T.reshape((96748,1)),
                         PCL_nor.normals),axis = 1)

cloud_with_normals = pcl.PointCloud.PointXYZINormal(Points)

ms = pcl.surface.GreedyProjectionTriangulation.PointNormal

M_PI = 3.14
triangles = pcl.PolygonMesh()
ms.setSearchRadius = 0.005
ms.setMu = 2.5
ms.setMaximumNearestNeighbors = 100
ms.setMaximumSurfaceAngle = M_PI/4
ms.setMinimumAngle =M_PI/18
ms.setMaximumAngle = 2*M_PI/3
ms.setNormalConsistency = True
ms.setInputCloud = cloud_with_normals
tree2 = pcl.search.KdTree.PointNormal()
ms.setSearchMethod = tree2
triangles = ms.reconstruct

pcl.io.PLYWriter.writeBinary ('Canopy.ply', triangles) 

I am not sure I am doing it in the right way. I did not manage to visualize or save the result (mesh).
Thanks in advance for your help.

Shouyang

lsymuyu, 2018-08-24:
Thanks for upgrading the library. Now it seems the triangulation works.
ms = pcl.surface.GreedyProjectionTriangulation.PointNormal()

M_PI = 3.14
triangles = pcl.PolygonMesh()
ms.setSearchRadius(0.005)
ms.setMu(2.5)
ms.setMaximumNearestNeighbors(100)
ms.setMaximumSurfaceAngle(M_PI/4)
ms.setMinimumAngle(M_PI/18)
ms.setMaximumAngle(2*M_PI/3)
ms.setNormalConsistency(True)
ms.setInputCloud(PCL_nor)
tree2 = pcl.search.KdTree.PointNormal()
ms.setSearchMethod(tree2)
ms.reconstruct(triangles)

Now the question is to save and visualize the triangles (Polygon mesh format). Thanks for your help.

pcl.io.PLYWriter.write('gg.ply',triangles.cloud) not working...

Shouyang

davidcaron, 2018-08-24:
I looked into it, and the function you would need is `pcl::io::savePLYFile`.

It looks like this function, along with a bunch more were skipped because their namespace was parsed differently... I tried to add them, but it creates a lot of small issues that I would need some more time to fix.

I'll do it when I have more time.

davidcaron, 2018-08-24:
Try v0.11.0, it should work.

```python
triangles = pcl.PolygonMesh()
# ...
ms.reconstruct(triangles)
pcl.io.savePLYFile("output.ply", triangles)
```

closed 2018-08-24 by davidcaron

lsymuyu, 2018-08-29:
Thanks for your nice improvement.

However, I found the visualization of polygonmesh and also point cloud is not working. 
The window can pop-up but crash afterwards.
viewer = pcl.visualization.PCLVisualizer("viewer")
viewer.setBackgroundColor(0, 0, 0)
viewer.viewer.addPolygonMesh(triangles)
Thanks again.
Best
SY

davidcaron, 2018-08-29:
Try adding this code. This starts the main loop of the VTK window.

```python 
while not viewer.wasStopped():
    v.spinOnce(50)
```

davidcaron, 2018-08-29:
But I never tried it with a mesh so post your results :)

AndreAhmed, 2019-02-06:
TypeError: addPolygonMesh(): incompatible function arguments. The following argument types are supported:
    1. (self: pclpy.pcl.visualization.PCLVisualizer, cloud: pclpy.pcl.PointCloud.PointSurfel, vertices: pclpy.pcl.vectors.Vertices, id: str = 'polygon', viewport: int = 0) -> bool
    2. (self: pclpy.pcl.visualization.PCLVisualizer, cloud: pclpy.pcl.PointCloud.PointXYZ, vertices: pclpy.pcl.vectors.Vertices, id: str = 'polygon', viewport: int = 0) -> bool
    3. (self: pclpy.pcl.visualization.PCLVisualizer, cloud: pclpy.pcl.PointCloud.PointXYZL, vertices: pclpy.pcl.vectors.Vertices, id: str = 'polygon', viewport: int = 0) -> bool
    4. (self: pclpy.pcl.visualization.PCLVisualizer, cloud: pclpy.pcl.PointCloud.PointXYZI, vertices: pclpy.pcl.vectors.Vertices, id: str = 'polygon', viewport: int = 0) -> bool
    5. (self: pclpy.pcl.visualization.PCLVisualizer, cloud: pclpy.pcl.PointCloud.PointXYZRGB, vertices: pclpy.pcl.vectors.Vertices, id: str = 'polygon', viewport: int = 0) -> bool
    6. (self: pclpy.pcl.visualization.PCLVisualizer, cloud: pclpy.pcl.PointCloud.PointXYZRGBA, vertices: pclpy.pcl.vectors.Vertices, id: str = 'polygon', viewport: int = 0) -> bool
    7. (self: pclpy.pcl.visualization.PCLVisualizer, cloud: pclpy.pcl.PointCloud.PointNormal, vertices: pclpy.pcl.vectors.Vertices, id: str = 'polygon', viewport: int = 0) -> bool
    8. (self: pclpy.pcl.visualization.PCLVisualizer, cloud: pclpy.pcl.PointCloud.PointXYZRGBNormal, vertices: pclpy.pcl.vectors.Vertices, id: str = 'polygon', viewport: int = 0) -> bool
    9. (self: pclpy.pcl.visualization.PCLVisualizer, cloud: pclpy.pcl.PointCloud.PointXYZRGBL, vertices: pclpy.pcl.vectors.Vertices, id: str = 'polygon', viewport: int = 0) -> bool
    10. (self: pclpy.pcl.visualization.PCLVisualizer, cloud: pclpy.pcl.PointCloud.PointWithRange, vertices: pclpy.pcl.vectors.Vertices, id: str = 'polygon', viewport: int = 0) -> bool

AndreAhmed, 2019-02-06:
triangles = pcl.PolygonMesh()
ms.setSearchRadius(0.2)
ms.setMu(2.5)
ms.setMaximumNearestNeighbors(100)
ms.setMaximumSurfaceAngle(pi / 4)
ms.setMinimumAngle(pi / 18)
ms.setMaximumAngle(2 * pi / 3)
ms.setNormalConsistency(True)
ms.setInputCloud(cloud_with_normals)
tree2 = pcl.search.KdTree.PointNormal()
ms.setSearchMethod(tree2)
ms.reconstruct(triangles)

viewer = pcl.visualization.PCLVisualizer("viewer")
viewer.setBackgroundColor(0, 0, 0)
viewer.addPolygonMesh(ms)
while not viewer.wasStopped():
    viewer.spinOnce(50)