davidcaron/pclpy #21: PCLVisualizer not working in pclpy

pankaj203kumar, 2018-09-03:
Hi, I am trying to use PCL library after installing pclpy and its dependencies packages. But when I use pcl.visualization.PCLVisualizer command for visualization purpose then new window pops up but then it gets struck and doesn't show anything. Can anyone help me with this issue ? I have provided few line of codes which I am using and attached the snapshot of new window getting hanged after using these codes. Thanks !!!

viewer = pcl.visualization.PCLVisualizer("3D Viewer")
viewer.setBackgroundColor(1, 0, 0)
viewer.addPointCloud(my_file)
viewer.addCoordinateSystem(1.0)
viewer.initCameraParameters()
viewer.resetCamera()_

niranjanreddy891, 2018-09-06:
@pankaj203kumar, try this

```
def test_pcl_visualizer_simple():
    pc = pclpy.io.read(test_data("file.las"), "PointXYZRGBA")
    viewer = pcl.visualization.PCLVisualizer("viewer", False)
    viewer.setBackgroundColor(0, 0, 0)
    rgb = pcl.visualization.PointCloudColorHandlerRGBAField.PointXYZRGBA(pc)
    viewer.addPointCloud(pc, rgb, "sample cloud")
    viewer.setPointCloudRenderingProperties(0, 3, "sample cloud")
    viewer.addCoordinateSystem(1.0)
    viewer.initCameraParameters()
    viewer.resetCamera()
```

pankaj203kumar, 2018-09-06:
Hi Niranjan,

thanks for your advice. I tried this thing in a way of calling the function but its again same issue. I think there might be some problem with PCLVisualizer wrapper while calling C++ functionality. Any other suggestion will also be welcome.

Regards

niranjanreddy891, 2018-09-06:
```
def test():
    pc = pclpy.io.read(test_data("street.las"), "PointXYZRGBA")
    viewer = pcl.visualization.PCLVisualizer("viewer", False)
    viewer.addPointCloud(pc)
    viewer.resetCamera()

viewers = [viewer]
```

davidcaron, 2018-09-07:
Hi, just like in c++, the viewer needs to be rendered inside a loop. Here is a code sample:

``` python
pc = pclpy.io.read("your_cloud.las", "PointXYZRGBA")
viewer = pcl.visualization.PCLVisualizer("viewer")
viewer.setBackgroundColor(0, 0, 0)
rgb = pcl.visualization.PointCloudColorHandlerRGBAField.PointXYZRGBA(pc)
viewer.addPointCloud(pc, rgb, "sample cloud")
viewer.setPointCloudRenderingProperties(0, 3, "sample cloud")
viewer.addCoordinateSystem(1.0)
viewer.resetCamera()

# this is the part you were missing
while not viewer.wasStopped():
    viewer.spinOnce(10)
```

Or, you could simply do:

``` python
pc = pclpy.read("your_cloud.las", "PointXYZRGBA")
pc.show()
```

pankaj203kumar, 2018-09-13:
Hi David,

Yes, I think that was a part which I missed in my code. Many thanks for sorting out this problem.

And we really appreciate your efforts in providing this wrapper of PCL in python. Thanks again for this !!!

Regards

closed 2018-09-13 by pankaj203kumar