davidcaron/pclpy #24: build on OSX

toinsson, 2018-09-26:
Hello, 

I have been trying to get a build on OSX. Did change a few files but got stop at this error:
`pclpy/src/make_opaque_vectors.hpp:18:34: error: no template named 'Vertices' in namespace 'pcl'; did you mean 'boost::vertices'?
`
Any idea?

toinsson, 2018-09-26:
adding `#include <pcl/Vertices.h>` to `make_opaque_vectors.hpp` solves the issue.

toinsson, 2018-09-26:
get to that point:
`./src/generated_modules/sample_consensus/sac_model.hpp:19:11: error: name defined in alias declaration must be an identifier
    using typename Class = pcl::Functor<_Scalar>;
`

davidcaron, 2018-09-26:
Thank you for working on this.
Compiling the whole thing is very long (~1 hour on a good pc), try to cache some modules or compile small parts at a time.
I'll try to help you as best as I can, but I don't have access to OSX myself.
It looks like the issue is at line 19 of `sac_model.hpp`? Could you copy/paste it?

toinsson, 2018-09-27:
Hello, 
I have pushed a commit to my fork in case you want to see the changes I had to implement for getting through the first compilation errors:
- some hacky changes in setup.py for finding the libs
- changes in includes
- changes in pclpy.cpp from module to object - might introduce a bug?
- adding typename in class definitions (contructor and var)
- make_opaque uses define (https://stackoverflow.com/questions/38030048/too-many-arguments-provided-to-function-like-macro-invocation)

I'm still stuck with these 2 which happen in several places:
1.
`
./src/generated_modules/features/board.hpp:17:68: error: unknown type name 'ReferenceFrame'; did you mean 'pcl::ReferenceFrame'?
template<typename PointInT, typename PointNT, typename PointOutT = ReferenceFrame>
                                                                   ^~~~~~~~~~~~~~
`
2.
`
In file included from ./src/generated_modules/_features_loader.cpp:30:
./src/generated_modules/features/integral_image_normal.hpp:38:15: error: template argument for template type parameter must be a type; did you forget
      'typename'?
    py::enum_<Class::BorderPolicy>(cls, "BorderPolicy")
              ^
              typename 
`

and this one, which I don't know how to fix:
`
In file included from ./src/generated_modules/_features_loader.cpp:2:
In file included from /Users/antoine/anaconda/envs/pclpy/include/python3.7m/pybind11/pybind11.h:43:
In file included from /Users/antoine/anaconda/envs/pclpy/include/python3.7m/pybind11/detail/../attr.h:13:
/Users/antoine/anaconda/envs/pclpy/include/python3.7m/pybind11/cast.h:1382:5: error: static_assert failed due to requirement 'std::is_base_of<base,
      type_caster<vector<Matrix<float, -1, -1, 0, -1, -1>, aligned_allocator<Matrix<float, -1, -1, 0, -1, -1> > > > >::value'
      "Holder classes are only supported for custom types"
    static_assert(std::is_base_of<base, type_caster<type>>::value,
    ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
`
has to do with:
`PYBIND11_DECLARE_HOLDER_TYPE(T, boost::shared_ptr<T>);
`

I won't have much time to continue working on this unfortunately, but will check potentially from time to time. 
Hope this helps. Cool project.