davidcaron/pclpy #29: 3d sift features extraction

chaiban, 2018-11-30:
Sorry this is not really an issue. But i have been seaching for a pcl wrapper for python. 

Does this pcl wrapper can find the sift features ? If possible please send a snippet.
I tested with python-pcl wrapper but SIFT feature library for python is missing. 

I really hope your wrapper works.

Thank you

davidcaron, 2018-11-30:
Unfortunately no, the `keypoints` module is not implemented but I think it wouldn't be that much work... I'll leave this here as a reminder for a future version.

chaiban, 2018-11-30:
Thank you for the information

justus-hildebrand, 2021-02-16:
The readme claims the `keypoints` and `features` modules are supported now, so is this issue still ongoing? I'm looking to use PCL from python myself for pointcloud feature extraction in python, pclpy looks promising if these modules work now!