davidcaron/pclpy #31: invalid template Error when attempting to build

adcrossan1, 2019-01-15:
Hi,
I am having an issue with templates when trying to build. I am using python 3.6 (anaconda), windows 10, pcl 1.8.1 , vs 2017. 
When  i run the  `python setup.py install` command, the following is displayed.

`(py_36) C:\Users\v1059\Downloads\pclpy-master\pclpy-master>python setup.py install
running install
running bdist_egg
running egg_info
writing pclpy.egg-info\PKG-INFO
writing dependency_links to pclpy.egg-info\dependency_links.txt
writing requirements to pclpy.egg-info\requires.txt
writing top-level names to pclpy.egg-info\top_level.txt
reading manifest file 'pclpy.egg-info\SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'pclpy.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
copying pclpy\src\generated_modules\_2d_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_base_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_common_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_features_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_filters_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_geometry_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_io_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_kdtree_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_octree_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_recognition_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_sample_consensus_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_search_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_segmentation_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_stereo_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_surface_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_tracking_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
copying pclpy\src\generated_modules\_visualization_loader.cpp -> build\lib.win-amd64-3.6\pclpy\src\generated_modules
running build_ext
building 'pclpy.pcl' extension
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET=1 -IC:\Users\v1059\Downloads\InstalledApps\Anaconda\envs\py_36\lib\site-packages\numpy\core\include "-IC:\Program Files\PCL 1.8.1\include\pcl-1.8" "-IC:\Program Files\PCL 1.8.1\3rdParty\Eigen\eigen3" "-IC:\Program Files\PCL 1.8.1\3rdParty\Boost\include\boost-1_64" "-IC:\Program Files\PCL 1.8.1\3rdParty\FLANN\include" "-IC:\Program Files\PCL 1.8.1\3rdParty\Qhull\include" -IC:\Users\v1059\Downloads\InstalledApps\Anaconda\envs\py_36\Library\include\vtk-8.1 -IC:\Users\v1059\Downloads\InstalledApps\Anaconda\envs\py_36\Library\include -IC:\Users\v1059\Downloads\InstalledApps\Anaconda\envs\py_36\Include -IC:\Users\v1059\AppData\Roaming\Python\Python36\Include -IC:\Users\v1059\Downloads\InstalledApps\Anaconda\envs\py_36\include -IC:\Users\v1059\Downloads\InstalledApps\Anaconda\envs\py_36\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" /EHsc /Tppclpy/src/pclpy.cpp /Fobuild\temp.win-amd64-3.6\Release\pclpy/src/pclpy.obj /EHsc /openmp /DVERSION_INFO=\"0.11.0\"
pclpy.cpp
Unknown compiler version - please run the configure tests and report the results
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(18): error C2039: 'Vertices': is not a member of 'pcl'
C:\Program Files\PCL 1.8.1\include\pcl-1.8\pcl/segmentation/supervoxel_clustering.h(59): note: see declaration of 'pcl'
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(18): error C2065: 'Vertices': undeclared identifier
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(18): error C2923: 'std::vector': 'Vertices' is not a valid template type argument for parameter '_Ty'
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(18): error C3203: 'allocator': unspecialized class template can't be used as a template argument for template parameter '_Alloc', expected a real type
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(18): error C3203: 'vector': unspecialized class template can't be used as a template argument for template parameter 'type', expected a real type
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(25): warning C4002: too many arguments for function-like macro invocation 'PYBIND11_MAKE_OPAQUE'
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(25): error C2059: syntax error: ':'
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(25): error C2143: syntax error: missing '>' before ';'
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(25): error C2974: 'pybind11::detail::type_caster_base': invalid template argument for 'type', type expected
c:\users\v1059\downloads\installedapps\anaconda\envs\py_36\include\pybind11\cast.h(778): note: see declaration of 'pybind11::detail::type_caster_base'
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(26): warning C4002: too many arguments for function-like macro invocation 'PYBIND11_MAKE_OPAQUE'
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(26): error C2059: syntax error: ':'
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(26): error C2143: syntax error: missing '>' before ';'
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(26): error C2974: 'pybind11::detail::type_caster_base': invalid template argument for 'type', type expected
c:\users\v1059\downloads\installedapps\anaconda\envs\py_36\include\pybind11\cast.h(778): note: see declaration of 'pybind11::detail::type_caster_base'`

I have created multiple fresh environments and am getting the same error each time. I was hoping for some help? I am quite new to python and any help would be appreciated.

davidcaron, 2019-01-15:
Hi, did you follow the instructions in `Readme.md`? 

Specifically, I think this could be related with the pybind11 version. You should try installing it from github if you didn't already `pip install git+https://github.com/pybind/pybind11` It _could_ be added in pybind11 version 2.2.4 but I'm not sure.

adcrossan1, 2019-01-16:
Thanks. Installing straight from github removed a lot of the errors however i'm still left with these if you had any idea as to whats causing them?

`pclpy.cpp
Unknown compiler version - please run the configure tests and report the results
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(18): error C2039: 'Vertices': is not a member of 'pcl'
C:\Program Files\PCL 1.8.1\include\pcl-1.8\pcl/segmentation/supervoxel_clustering.h(59): note: see declaration of 'pcl'
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(18): error C2065: 'Vertices': undeclared identifier
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(18): error C2923: 'std::vector': 'Vertices' is not a valid template type argument for parameter '_Ty'
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(18): error C3203: 'allocator': unspecialized class template can't be used as a template argument for template parameter '_Alloc', expected a real type
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\make_opaque_vectors.hpp(18): error C3203: 'vector': unspecialized class template can't be used as a template argument for template parameter 'type', expected a real type
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\vector_classes.hpp(30): error C2039: 'Vertices': is not a member of 'pcl'
C:\Program Files\PCL 1.8.1\include\pcl-1.8\pcl/segmentation/supervoxel_clustering.h(59): note: see declaration of 'pcl'
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\vector_classes.hpp(30): error C2065: 'Vertices': undeclared identifier
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\vector_classes.hpp(30): error C2923: 'std::vector': 'Vertices' is not a valid template type argument for parameter '_Ty'
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\vector_classes.hpp(30): error C3203: 'allocator': unspecialized class template can't be used as a template argument for template parameter '_Alloc', expected a real type
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\vector_classes.hpp(30): error C2672: 'bindVector': no matching overloaded function found
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\vector_classes.hpp(30): error C3206: 'bindVector': invalid template argument for 'T', missing template argument list on class template 'std::vector'
c:\users\v1059\downloads\pclpy-master\pclpy-master\pclpy\src\vector_classes.hpp(17): note: see declaration of 'bindVector'
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX64\\x64\\cl.exe' failed with exit status 2
`