davidcaron/pclpy #34: pip install requirements fails

barlaensdoonn, 2019-02-17:
installing requirements with `pip install -r requirements.txt`
fails with:

```
  Could not find a version that satisfies the requirement pclpy_dependencies>=0.2.1 (from -r requirements.txt (line 3)) (from versions: )
No matching distribution found for pclpy_dependencies>=0.2.1 (from -r requirements.txt (line 3))
```

similarly `pip install pclpy` fails with:
```
  Could not find a version that satisfies the requirement pclpy (from versions: )
No matching distribution found for pclpy
```

also `pip install -r requirements_dev.txt` fails with:
```
  Could not find a version that satisfies the requirement pybind11>=2.3dev (from -r requirements_dev.txt (line 1)) (from versions: 1.0, 1.2, 1.3, 1.4, 1.5, 1.7, 1.8, 1.8.1, 2.0.0rc1, 2.0.0, 2.0.1, 2.1.0, 2.1.1, 2.2.0, 2.2.1, 2.2.2, 2.2.3, 2.2.4)
No matching distribution found for pybind11>=2.3dev (from -r requirements_dev.txt (line 1))
```

Sajjadmanal, 2019-03-05:
Hi, I am having the similar issue. Unable to install it.

niranjanreddy891, 2019-03-19:
@barlaensdoonn and @Sajjadmanal could you please specify the version of Python and O.S?

barlaensdoonn, 2019-03-20:
ubuntu 16.04 and python 3.5.2

niranjanreddy891, 2019-05-10:
@barlaensdoonn currently pclpy doesn't support for Ubuntu platform.

mrinalsenapati04, 2019-07-27:
facing the same issue in anaconda navigator on windows 10 ... my python version is  3.7

MengHao666, 2020-05-07:
> @barlaensdoonn currently pclpy doesn't support for Ubuntu platform.
and what happens now as for this issue?

davidcaron, 2020-05-07:
The current version switched to the conda packaging platform and works on Ubuntu. I'm closing this issue as there's no intent on supporting pypi in the future. Please open another issue if there's a problem.

closed 2020-05-07 by davidcaron