davidcaron/pclpy #36: Setup in Linux using Anaconda Enviroment

SolidusAbi, 2019-03-20:
Im trying to prepare the pclpy in order to work in a anaconda enviroment on Linux but I have problem with CppHeaderParser in the ~/pcl/octree/octree_iterator.h file.

This is the error:
CppHeaderParser.CppHeaderParser.CppParseError: Not able to parse  on line 78 evaluating "{"
Error around: class OctreeIteratorBase : public std : : iterator <std : : forward_iterator_tag , const OctreeNode , void , const OctreeNode * , const OctreeNode & >

I don't understand what is the problem, does it happen in Windows?  Or does someone have a similar problem?

Thank you.

davidcaron, 2019-03-20:
Hi,

Linux is not supported at the moment... 

But the error you're getting would probably be fixed if you used github.com/davidcaron/cppheaderparser

I changed some things in cppheaderparser and this repo is not on github so I didn't submit a pr...

SolidusAbi, 2019-03-21:
Thank you, your CppHeaderParser works great. For your interest, in PCL 1.9 the EuclideanClusterComparator file (euclidian_cluster_comparator.h) contains two class with the same name, in different namespaces, and the parser crashes but in 1.8 everything works great . 

Anyway, I know there is no support for Linux, Im trying to contribute because I think your project is interesting and python-pcl doesn't work very well. My idea is to generate an anaconda repository. 

I'll leave this issue open until I get to finish. 

Thank you

davidcaron, 2019-03-21:
If you want to contribute that's great!

I just pushed some commits that I did a while ago in this branch: https://github.com/davidcaron/pclpy/tree/linux-compilation

I suggest you start by checking this. It's not done, but I already spent some time on it, so you can check it out.

geirbi, 2019-06-13:
@SolidusAbi @davidcaron Is there any progress on this? I'd really like to have pclpy support on Linux.

SolidusAbi, 2019-06-14:
@geirbi I'm sorry but I haven't had time to continue, I'm also very interested but I think it's going to take quite some time.