davidcaron/pclpy #39: C++ version to python version，std::multimap<uint32_t, uint32_t>

millionsofluo, 2019-04-18:
Hi，

how to get `std::multimap<uint32_t, uint32_t> supervoxel_adjacency` data type input supervox.getSupervoxelAdjacency()

C++ code：
` std::multimap<uint32_t, uint32_t> supervoxel_adjacency;`
 `super.getSupervoxelAdjacency(supervoxel_adjacency);`

question：
TypeError: getSupervoxelAdjacency(): incompatible function arguments. The following argument types are supported:
    1. (self: pclpy.pcl.segmentation.SupervoxelClustering.PointXYZRGBA, label_adjacency: std::multimap<unsigned int,unsigned int,std::less<unsigned int>,std::allocator<std::pair<unsigned int const ,unsigned int> > >) -> None

c++
```
      float voxel_resolution = 0.008f;
      float seed_resolution = 0.1f;
      float color_importance = 0.2f;
      float spatial_importance = 0.4f;
      float normal_importance = 1.0f;
      pcl::SupervoxelClustering<PointT> super (voxel_resolution, seed_resolution);
      super.setInputCloud (cloud);
      super.setColorImportance (color_importance);
      super.setSpatialImportance (spatial_importance);
      super.setNormalImportance (normal_importance);
      std::map <uint32_t, pcl::Supervoxel<PointT>::Ptr > supervoxel_clusters;
      super.extract (supervoxel_clusters);
      std::multimap<uint32_t, uint32_t> supervoxel_adjacency;
      super.getSupervoxelAdjacency (supervoxel_adjacency);
```
my python code
```
    voxel_resolution = 1
    seed_resolution = 2
    color_importance = 2
    spatial_importance = 4
    normal_importance = 1.0
    supervox = pcl.segmentation.SupervoxelClustering.PointXYZRGBA(voxel_resolution, seed_resolution)
    supervox.setInputCloud(cloud)
    supervox.setColorImportance(color_importance)
    supervox.setSpatialImportance(spatial_importance)
    supervox.setNormalImportance(normal_importance)
    supervoxel_clusters = pcl.vectors.map_uint32t_PointXYZRGBA()
    supervox.extract(supervoxel_clusters)
    supervoxel_adjacency = ???
    supervox.getSupervoxelAdjacency(supervoxel_adjacency)

```

please help
Thanks!

Famok, 2019-05-15:
Hi,

Same problem here! This would be incredible useful.

davidcaron, 2019-05-15:
The `std::multimap<uint32_t, uint32_t>` type would need to be explicitly added to `pclpy/src/vector_classes.hpp` it's not being compiled currently. And a quick search for multimap in pybind11's documentation doesn't return anything... so that could be an issue as well.

Famok, 2019-05-15:
So there is no way to integrate this anytime soon?

millionsofluo, 2019-05-22:
> The `std::multimap<uint32_t, uint32_t>` type would need to be explicitly added to `pclpy/src/vector_classes.hpp` it's not being compiled currently. And a quick search for multimap in pybind11's documentation doesn't return anything... so that could be an issue as well.

Thank you. I'll try.