davidcaron/pclpy #42: conditional_removal

Famok, 2019-05-15:
Hi,

Are there any plans to include conditional_removal?
Right now it says in "what to skip"
("filters", "conditional_removal.h"),
# todo: parser error for ConditionalRemoval (int extract_removed_indices = false)

davidcaron, 2019-05-15:
It seems to be relatively simple to add... From what I remember, I think this was due to a CppHeaderParser error. 

In the meantime, I could be wrong but maybe the same functionality could be achieved using numpy functions? if you have a XYZ point cloud, I think you can do something like:

``` python
xyz = point_cloud.xyz  # xyz is an array of shape (n_points, 3)
keep_above_z_value = 10
filtered_cloud = xyz[xyz[:, 2] > keep_above_z_value]
```

Famok, 2019-05-22:
I've misunderstood the use of conditional removal. You are right, it can be done with numpy!

closed 2019-05-22 by Famok