davidcaron/pclpy #48: pip install pypi

WillingDil, 2019-09-25:
安装成功之后   import pclpy 提示下面的错误

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "D:\Python36\lib\site-packages\pclpy\__init__.py", line 3, in <module>
    import pclpy.pcl as pcl
ImportError: DLL load failed: 找不到指定的模块。

niranjanreddy891, 2019-09-25:
Install OpenNI2, it will solve the issue. Make sure you copy C:\Program Files\OpenNI2\Samples\Bin\OpenNI2.dll.Copy OpenNI2.dll to Lib\site-packages\python_pcl-0.3-py3.6-win-amd64.egg\pcl

lfhgljj, 2019-11-25:
I also have the same problem.but there is no "python_pcl-0.3-py3.6-win-amd64.egg" in “site-packages”.The same situation occurred when I used “pip” to install it.
I installed it directly using pycharm. Pycharm uses the local environment. The computer has installed pcl(1.8.1，2015). By the way , I used python3.6.
How can I fix it?

lfhgljj, 2019-11-25:
After a day of hard work, I have I finally solved the problem。After trying various methods, I used a relatively silly method. I deleted the ".PyCharm2019.2" folder and reimported the local interpreter.
Although the problem was solved, the reason was still unclear. If anyone knows the reason, please help me explain the reason when you are free. Grateful!

danielemarchisio, 2020-01-14:
I had the same problem, but after some time, I decided to copy all .dll files from pclpy_dependencies to pclpy folder and this worked for me! Don't know why

bernland, 2020-02-19:
I experienced the exact same problem and solved it by recovering the file "C:\WINDOWS\System32\msvcr100.dll" from another system.

I hope this helps!

CRIGIM, 2020-06-16:
I have the same issue. None of the proposed solution is working for me

ankus-wu, 2020-08-31:
you may need to install c++ tools in Visual Studio Installer

Alaskayoo, 2023-01-14:
> I had the same problem, but after some time, I decided to copy all .dll files from pclpy_dependencies to pclpy folder and this worked for me! Don't know why

I did this and  it works for pclpy. BUT then it went wrong with import pcl. Is there a conflict between them? They don't seem to work at the same time.