davidcaron/pclpy #53: Possible to read .pcd files?

RazanulHoque, 2020-03-01:
Hi! 

Forgive my intrusion! I'm very new here, but I was wondering if there was any possibility to read .pcd files with pclpy? Because so far, as I understand it, I've only found out that only .las files are readable? I'm currently doing a project for my university, and I'm working with mostly .pcd files.

zhchyang2004, 2020-05-12:
Hi, RazanulHoque, please follow this:

from pclpy import pcl

filename='../chef.pcd'
obj=pcl.PointCloud.PointXYZ()

#Option 1:
pcl.io.loadPCDFile(filename,obj)
#Option 2:
pcl.io.PCDReader().read(filename,obj)

obj.show()

Mayank756, 2020-07-22:
Tried doing the same but it's giving the following error:

'pclpy.pcl.PointCloud.PointXYZ' object has no attribute 'show'

Any Idea how can I solve it?