davidcaron/pclpy #55: ImportError: DLL load failed: %1

zeken959, 2020-03-20:
Traceback (most recent call last):
  File "C:/Users/86132/PycharmProjects/biye/bbbbb.py", line 2, in <module>
    import pclpy
  File "C:\Users\86132\PycharmProjects\biye\venv\lib\site-packages\pclpy\__init__.py", line 3, in <module>
    import pclpy.pcl as pcl
ImportError: DLL load failed: %1 不是有效的 Win32 应用程序。

JinXiangLai, 2020-03-29:
Try: from pclpy import pcl instead of import pclpy.pcl as pcl

laiming997, 2020-04-12:
> Traceback (most recent call last):
> File "C:/Users/86132/PycharmProjects/biye/bbbbb.py", line 2, in
> import pclpy
> File "C:\Users\86132\PycharmProjects\biye\venv\lib\site-packages\pclpy__init__.py", line 3, in
> import pclpy.pcl as pcl
> ImportError: DLL load failed: %1 不是有效的 Win32 应用程序。

请问你解决问题了吗

koalaaaaaaaaa, 2022-03-28:
vscode没法导入这样报错，但是命令行可以，很神奇