davidcaron/pclpy #56: No keypoint extractors located

JohanThrane, 2020-03-26:
Hello

I've been going through you library trying to find a keypoint extractor. I know they exist in the original pcl library, but i was unable to locate any in pclpy. It appears that a module for keypoints was supposed to be built, when i look at pclpy/generators/config.py, but that module does not appear when i run

print(dir(pclpy.pcl))

or when i run it on subsequent modules. Have you not yet implemented keypoint detectors or have i just overlooked them?

Thank you very much for your work and your help.

stefangachter, 2020-05-01:
Hi David,
I am looking for the keypoint extractors as well. I cannot find them and I am unsure how to access them.
I try the following
`narf = pcl.features.NarfDescriptor.PointWithRange_Narf36()`
but without success.
I would appreciate your help. Thanks!

13117532, 2022-08-25:
Hey there,
I could still not find keypoint extractors. Do they not exist, or have I just overlooked them?

Thanks for all the help.