davidcaron/pclpy #59: I get something wrong with the function 'compute3DCentroid'

laiming997, 2020-04-11:
Thank you for your work of the pcl to python. And when I use the pclpy to compute the point cloud centroid and compute Covariance Matrix. I was confused about the result I get. I initialize a array which is [[1.0],[1.0],[1.0],[1.0]], and it's the array([4,1],float) variable，when I use this code `pcl.common.compute3DCentroid(cloud_filter,centroid)` to compute, the result is always  [[1.0],[1.0],[1.0],[1.0]], it has no change , and the function result I get is the size of the pointCloud which is mean the Centroid has no change. But when I use this code `pcl.common.computeCentroid(cloud_filter,centroid_point)` , I get the normal result . Is there any problem of my usage? How can I compute the Centroid correctly?

laiming997, 2020-04-11:
@davidcaron

Marvelsh, 2020-05-26:
I can't find the function pcl.common.compute3DCentroid in pclpy

naah69, 2020-07-16:
> Thank you for your work of the pcl to python. And when I use the pclpy to compute the point cloud centroid and compute Covariance Matrix. I was confused about the result I get. I initialize a array [...]

so ami i,have u detail it

davidcaron, 2020-07-16:
You're right that the `compute3DCentroid` function is not there.

For this case, I think I would suggest using numpy. You can get a numpy view of the points (no memory is copied) by using the `.xyz` property:

``` python
>>> import pclpy
>>> import numpy as np
>>> pc = pclpy.pcl.PointCloud.PointXYZ.from_array([[1, 0, 0], [0, 1, 0], [0, 0, 1], [1, 1, 1]])
>>> np.cov(pc.xyz.T)
array([[0.33333333, 0.        , 0.        ],
       [0.        , 0.33333333, 0.        ],
       [0.        , 0.        , 0.33333333]])
>>> centroid = pc.xyz.mean(axis=0)
>>> centroid
array([0.5, 0.5, 0.5], dtype=float32)
```

naah69, 2020-07-16:
is there `computeCovarianceMatrixNormalized`? it didn't change too

davidcaron, 2020-07-16:
At first glance, I'm not sure why some on these functions didn't get compiled. They probably have arguments or outputs that are tricky to bind to python.

naah69, 2020-07-16:
oh，It's really a sad story

naah69, 2020-07-16:
i found that both of `compute3DCentroid ` and `computeCovarianceMatrixNormalized `  are use numpy array as param.

are the others same problem?