davidcaron/pclpy #62: ImportError: libflann.so.1.9: cannot open shared object file: No such file or directory

MengHao666, 2020-05-07:
I try to install it in ubuntu16.04, python3.6 now using "conda install -c conda-forge -c davidcaron pclpy"
  and now i got erros  like the following:
    "Python 3.6.10 |Anaconda, Inc.| (default, Mar 25 2020, 23:51:54) 
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pclpy
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/haomeng/anaconda3/envs/py36/lib/python3.6/site-packages/pclpy/__init__.py", line 3, in <module>
    import pclpy.pcl as pcl
ImportError: libflann.so.1.9: cannot open shared object file: No such file or directory
>>> 
"

How can i solve such problem? Should i build it from source？
Ｏr it cannot support ubuntu now?

MengHao666, 2020-05-08:
Ｄoes anyone else meet same problem?

wisper181, 2021-04-27:
hello,I get the same problem.Have U solved the BUG?
Any suggestion may be helpful
Thanks in advance

MengHao666, 2021-04-27:
I am sorry that I fogot it. ---- Replied Message ***@***.***>Date04/27/2021 16:09 ***@***.***> CcHao ***@***.******@***.***>SubjectRe: [davidcaron/pclpy] ImportError: libflann.so.1.9: cannot open shared object file: No such file or directory (#62)
hello,I get the same problem.Have U solved the BUG?
Any suggestion may be helpful
Thanks in advance

[
{
"target": "https://github.com/davidcaron/pclpy/issues/62#issuecomment-827408583",
"url": "https://github.com/davidcaron/pclpy/issues/62#issuecomment-827408583",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2021-04-27 by MengHao666