davidcaron/pclpy #63: conda install still incompatible on Ubuntu (18.04) and python 3.7

szx0112, 2020-05-13:
It seems that the recent version 0.12 supports the ubuntu and python3.7 as mentioned in the doc. However, when I still either through conda or pip or setup.py. I got the incomptaible problems.

Below is the conda installation:
 
```
conda install -c davidcaron pclpy
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: \ 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                            

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - pclpy -> python[version='>=3.6,<3.7.0a0|>=3.8,<3.9.0a0']

Your python: python=3.7

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.

The following specifications were found to be incompatible with your CUDA driver:

  - feature:/linux-64::__cuda==10.1=0
  - feature:|@/linux-64::__cuda==10.1=0

Your installed CUDA driver is: 10.1
```

szx0112, 2020-05-18:
Switched to python-pcl due to lack of documentations.

closed 2020-05-18 by szx0112

VisionaryMind, 2021-05-25:
> Switched to python-pcl due to lack of documentations.

Python-pcl does not implement OpenMP, which is one of the reasons this library is unique. Were you eventually able to get OMP working with pclpy? Python-pcl is no longer maintained and, for all intents and purposes, obsolete. I do not see any recent activity on this repository, so it's possible pclpy is also soon to be obsolete.

This brings up the question --- what do we do if we need to implement multi-processing in python-based pipelines? It looks to me that in the absence of a functional wrapper, we will have to resort to developing our own modules.