davidcaron/pclpy #67: writing outputs

Ciaran1981, 2020-06-23:
Thanks very much for this great lib!

More of a question than an issue - when processing something such as....

```
point_cloud.region_growing(norms, n_neighbours=30, min_size=100,
                           max_size=1000000, smooth_threshold=rad,
                           curvature_threshold=1, residual_threshold=0.1)
```
Is the output written to the object (point_cloud - I am guessing not)? If I return a variable I only have the indices. How is this written to the object?

khaoula-elh, 2020-06-27:
same question here !

Robin6416KW, 2023-07-17:
I try to do this recently,
I fallow this article but get some problem, 
https://stackoverflow.com/questions/59396391/how-to-get-a-specific-cluster-after-region-growing-using-pcl-library

the method is feeding the clusters into ExtractIndices then output the specific point.
my code is below:

----
pc = pclpy.read(inpcloud, "PointXYZI")
rg = pcl.segmentation.RegionGrowing.PointXYZI_Normal()
rg.setInputCloud(pc)
normals_estimation = pcl.features.NormalEstimationOMP.PointXYZI_Normal()
normals_estimation.setInputCloud(pc)
normals = pcl.PointCloud.Normal()
normals_estimation.setRadiusSearch(0.35)
normals_estimation.compute(normals)
rg.setInputNormals(normals)

rg.setMaxClusterSize(10)
rg.setMinClusterSize(5)
rg.setNumberOfNeighbours(15)
rg.setSmoothnessThreshold(5 / 180 * math.pi)
rg.setCurvatureThreshold(5)
rg.setResidualThreshold(1)
clusters = pcl.vectors.PointIndices()
rg.extract(clusters)

ext = pcl.filters.ExtractIndices.PointXYZI()
ext.setInputCloud(pc)
ext.setIndices(clusters[1])
ext.setNegative(False)
output_cloud = pcl.PointCloud.PointXYZI()
ext.filter(output_cloud)
----
but the ext.setIndices cloudn't accept the clusters that RegionGrowing's result.

in the article I share, they said that we need to transform clusters using this
extract.setIndices (pcl::PointIndicesConstPtr(clusters[cluster_idx]))

but in pclpy, I couldn't find a way to using "PointIndicesConstPtr" function.

this is my progress so far.