davidcaron/pclpy #70: pcl for Colab

wanderer799, 2020-10-02:
How to install this package on colab. So far its showing:

ERROR: Could not find a version that satisfies the requirement pclpy (from versions: none)
ERROR: No matching distribution found for pclpy

davidcaron, 2020-10-02:
Does colab support installing packages from conda?

wanderer799, 2020-10-02:
i think manually it si possible.I Used pip for windows as well as colab, this is the error i am getting.

Shanki5, 2020-11-22:
Unfortunately the pclpy is not available on pip. However you can install conda on colab and then use conda to install pclpy. 
[https://towardsdatascience.com/conda-google-colab-75f7c867a522](url) , gives a clear idea on how to install conda on colab and then proceeds to explain how to install packages using the same.

geothan, 2021-01-13:
Hello,

i'm trying to import pclpy in a google colab project, but i get:

---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-6-1e2e5cb12ae8> in <module>()
----> 1 import pcl

ModuleNotFoundError: No module named 'pcl'
----------------------------------------------------------------------------

I tried the following in order to setup the conda environment:

! wget https://repo.anaconda.com/miniconda/Miniconda3-py38_4.9.2-Linux-x86_64.sh
! chmod +x Miniconda3-py38_4.9.2-Linux-x86_64.sh
! bash ./Miniconda3-py38_4.9.2-Linux-x86_64.sh -b -f -p /usr/local
import sys
_ = (sys.path
        .append("/usr/local/lib/python3.6/site-packages"))

!conda install -c conda-forge -c davidcaron pclpy
import pcl

SilviaMarino, 2023-10-16:
Hello everyone, I'm having the same issue as @geothan, did you resolved it then? Is there any other way of using pclpy on colab?