davidcaron/pclpy #72: Building instrucation

adimesika1, 2020-11-02:
Hi, 
I have an issue with step 6 in the building part.
I have tried running both, developers and release and I got this error:
 
[image]

Thanks,
Adi.

davidcaron, 2020-11-02:
Hi, try to run `conda-build -c conda-forge .` from the `conda` folder instaed of the `scripts` folder. And as a reminder: the build is quite long, and 8Gb of ram is likely not enough.

adimesika1, 2020-11-02:
Hi, Thanks for the quick response.
I still got an error:
[image]