davidcaron/pclpy #76: Failed install PCL using conda Python 3.7 on Ubuntu 20.04

AlexisKerib, 2020-12-28:
As title
```
❯ python -V                                          
Python 3.7.9
```

```
❯ conda install -c conda-forge -c davidcaron pclpy   

Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: / 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                        

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - pclpy -> python[version='>=3.6,<3.7.0a0|>=3.8,<3.9.0a0']

Your python: python=3.7

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.

```
and i can't find linux-64/pclpy-0.12.0-py37_1.tar.bz2 here
https://anaconda.org/davidcaron/pclpy/files

davidcaron, 2020-12-28:
Hi, you're right that python 3.7 is not there :-/

You need to have python from conda-forge. To make sure of that, start from a clean environment:

`conda create -n my-environment`

`conda activate my-environment`

`conda install -c conda-forge -c davidcaron pclpy`

This should install python 3.8 with pclpy.

AlexisKerib, 2020-12-29:
You may make a change at readme :-\\
https://github.com/davidcaron/pclpy/blame/master/README.md#L45

tegusi, 2021-01-06:
Actually I created a new python 3.8.5 environment with anaconda,  
```
conda install --use-local pclpy-0.12.0-py38_1.tar.bz2
conda install -c local pclpy 
```
and use local installation, the error still exists that:
```
The environment is inconsistent, please check the package plan carefully
The following packages are causing the inconsistency:

  - <unknown>/linux-64::pclpy==0.12.0=py38_1                                                             failed with initial frozen solve. Retrying with flexible solve.
Solving environment: -
Found conflicts! Looking for incompatible packages.                                                      failed

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - pclpy -> python[version='>=3.6,<3.7.0a0|>=3.9,<3.10.0a0|>=3.7,<3.8.0a0']

Your python: python=3.8
```