davidcaron/pclpy #77: Failed to import pclpy using conda Python 3.8 on Google Colab

geothan, 2021-01-05:
Hello,

i'm trying to import pclpy in a google colab project, but i get:

---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-6-1e2e5cb12ae8> in <module>()
----> 1 import pcl

ModuleNotFoundError: No module named 'pcl'
----------------------------------------------------------------------------

I tried the following in order to setup the conda environment:

! wget https://repo.anaconda.com/miniconda/Miniconda3-py38_4.9.2-Linux-x86_64.sh
! chmod +x Miniconda3-py38_4.9.2-Linux-x86_64.sh
! bash ./Miniconda3-py38_4.9.2-Linux-x86_64.sh -b -f -p /usr/local
import sys
_ = (sys.path
        .append("/usr/local/lib/python3.6/site-packages"))

!conda install -c conda-forge -c davidcaron pclpy
import pcl

cyy280113999, 2021-04-07:
is that you use **import pcl** instead of **import pclpy** ?

SilviaMarino, 2023-10-16:
It doesn't work in both ways, at least for me. Has someone been able to import this binding in colab?