davidcaron/pclpy #90: Failed to import pclpy with conda Python 3.6 on CentOS 7.7 and Ubuntu 18.04

andrewfengusa, 2021-06-15:
Hi,

I got the following error when trying to import pclpy : 

_Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/centos/miniconda3/envs/STPLS_Centos/lib/python3.6/site-packages/pclpy/__init__.py", line 3, in <module>
    import pclpy.pcl as pcl
ImportError: libqhull_p.so.7: cannot open shared object file: No such file or directory_

I followed the instruction on the page and use "conda install -c conda-forge -c davidcaron pclpy". From the conda message it looks like the qhull is also part of the installation. Not sure why it's complaining about the missing .so. 

To make sure this is not a CentOS specific issue, I also tried the same setup on Ubuntu 18.04 and still got the same errors. Any guidance about how to fix this problem would be appreciated. Thanks.

Pilat66, 2021-07-08:
I confirm:

Ubuntu 20.04 ,
conda 4.10.3
Python 3.8.10
```
    import pclpy
  File "/home/reestr/miniconda3/envs/pdal/lib/python3.8/site-packages/pclpy/__init__.py", line 3, in <module>
    import pclpy.pcl as pcl
ImportError: libqhull_p.so.7: cannot open shared object file: No such file or directory
```

I3lacx, 2021-11-18:
As mentioned in #91 this worked for me:
`conda install -c conda-forge/label/gcc7 qhull`

Ubuntu 18.04
Python 3.8.12
conda 4.10.3