davidcaron/pclpy #91: Build error: ImportError: libqhull_p.so.7: cannot open shared object file: No such file or directory

Pilat66, 2021-07-09:
Hi!

Error 
`ImportError: libqhull_p.so.7: cannot open shared object file: No such file or directory`

Ubuntu 20.04 LTS
conda 4.10.3

```
user@host:~$ conda create -n pclpytest
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /home/user/miniconda3/envs/pclpytest

Proceed ([y]/n)? y

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
#
# To activate this environment, use
#
#     $ conda activate pclpytest
#
# To deactivate an active environment, use
#
#     $ conda deactivate

user@host:~$ conda activate pclpytest
(pclpytest) user@host:~$ conda install -c local pclpy 
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /home/user/miniconda3/envs/pclpytest

  added / updated specs:
    - pclpy

The following NEW packages will be INSTALLED:

  _libgcc_mutex      conda-forge/linux-64::_libgcc_mutex-0.1-conda_forge
  _openmp_mutex      conda-forge/linux-64::_openmp_mutex-4.5-1_gnu
  boost-cpp          conda-forge/linux-64::boost-cpp-1.72.0-h312852a_5
  bzip2              conda-forge/linux-64::bzip2-1.0.8-h7f98852_4
  c-ares             conda-forge/linux-64::c-ares-1.17.1-h7f98852_1
  ca-certificates    conda-forge/linux-64::ca-certificates-2021.5.30-ha878542_0
  certifi            conda-forge/linux-64::certifi-2021.5.30-py38h578d9bd_0
  eigen              conda-forge/linux-64::eigen-3.3.9-h4bd325d_1
  flann              conda-forge/linux-64::flann-1.9.1-h2e58136_1008
  hdf5               conda-forge/linux-64::hdf5-1.10.6-nompi_h6a2412b_1114
  icu                conda-forge/linux-64::icu-68.1-h58526e2_0
  krb5               conda-forge/linux-64::krb5-1.19.1-hcc1bbae_0
  laspy              conda-forge/noarch::laspy-2.0.1-pyh8a188c0_0
  ld_impl_linux-64   conda-forge/linux-64::ld_impl_linux-64-2.36.1-hea4e1c9_1
  libblas            conda-forge/linux-64::libblas-3.9.0-9_openblas
  libcblas           conda-forge/linux-64::libcblas-3.9.0-9_openblas
  libcurl            conda-forge/linux-64::libcurl-7.77.0-h2574ce0_0
  libedit            conda-forge/linux-64::libedit-3.1.20191231-he28a2e2_2
  libev              conda-forge/linux-64::libev-4.33-h516909a_1
  libffi             conda-forge/linux-64::libffi-3.3-h58526e2_2
  libgcc-ng          conda-forge/linux-64::libgcc-ng-9.3.0-h2828fa1_19
  libgfortran-ng     conda-forge/linux-64::libgfortran-ng-9.3.0-hff62375_19
  libgfortran5       conda-forge/linux-64::libgfortran5-9.3.0-hff62375_19
  libgomp            conda-forge/linux-64::libgomp-9.3.0-h2828fa1_19
  liblapack          conda-forge/linux-64::liblapack-3.9.0-9_openblas
  libnghttp2         conda-forge/linux-64::libnghttp2-1.43.0-h812cca2_0
  libopenblas        conda-forge/linux-64::libopenblas-0.3.15-pthreads_h8fe5266_1
  libssh2            conda-forge/linux-64::libssh2-1.9.0-ha56f1ee_6
  libstdcxx-ng       conda-forge/linux-64::libstdcxx-ng-9.3.0-h6de172a_19
  lz4-c              conda-forge/linux-64::lz4-c-1.9.3-h9c3ff4c_0
  ncurses            conda-forge/linux-64::ncurses-6.2-h58526e2_4
  numpy              conda-forge/linux-64::numpy-1.21.0-py38h9894fe3_0
  openssl            conda-forge/linux-64::openssl-1.1.1k-h7f98852_0
  pcl                conda-forge/linux-64::pcl-1.9.1-h2dfa329_1005
  pclpy              davidcaron/linux-64::pclpy-0.12.0-py38_1
  pip                conda-forge/noarch::pip-21.1.3-pyhd8ed1ab_0
  python             conda-forge/linux-64::python-3.8.10-h49503c6_1_cpython
  python_abi         conda-forge/linux-64::python_abi-3.8-2_cp38
  qhull              conda-forge/linux-64::qhull-2020.2-h4bd325d_1
  readline           conda-forge/linux-64::readline-8.1-h46c0cb4_0
  setuptools         conda-forge/linux-64::setuptools-49.6.0-py38h578d9bd_3
  sqlite             conda-forge/linux-64::sqlite-3.36.0-h9cd32fc_0
  tk                 conda-forge/linux-64::tk-8.6.10-h21135ba_1
  wheel              conda-forge/noarch::wheel-0.36.2-pyhd3deb0d_0
  xz                 conda-forge/linux-64::xz-5.2.5-h516909a_1
  zlib               conda-forge/linux-64::zlib-1.2.11-h516909a_1010
  zstd               conda-forge/linux-64::zstd-1.5.0-ha95c52a_0

Proceed ([y]/n)? y

Preparing transaction: done
Verifying transaction: done
Executing transaction: done

(pclpytest) user@host:~$ cd git
(pclpytest) user@host:~/git$ cd test-pcl/python/
(pclpytest) user@host:~/git/test-pcl/python$ ls
smooth.py

(pclpytest) user@host:~/git/test-pcl/python$ python smooth.py 
Traceback (most recent call last):
  File "smooth.py", line 3, in <module>
    import pclpy
  File "/home/user/miniconda3/envs/pclpytest/lib/python3.8/site-packages/pclpy/__init__.py", line 3, in <module>
    import pclpy.pcl as pcl
ImportError: libqhull_p.so.7: cannot open shared object file: No such file or directory

(pclpytest) user@host:~/git/test-pcl/python$ 
(pclpytest) user@host:~/git/test-pcl/python$ cat smooth.py

#!/bin/env python3

import pclpy
from pclpy import pcl

point_cloud = pclpy.read("Fragment.las", "PointXYZRGBA")
mls = pcl.surface.MovingLeastSquaresOMP.PointXYZRGBA_PointNormal()
tree = pcl.search.KdTree.PointXYZRGBA()
mls.setSearchRadius(0.05)
mls.setPolynomialFit(False)
mls.setNumberOfThreads(12)
mls.setInputCloud(point_cloud)
mls.setSearchMethod(tree)
mls.setComputeNormals(True)
output = pcl.PointCloud.PointNormal()
mls.process(output)(pclpytest) 

(pclpytest) user@host:
```

Pilat66, 2021-07-14:
From http://www.qhull.org/news/qhull-news.html:

Qhull's builds produce a shared library, libqhull_r.so (qhull_r.dll), several static libraries, and several applications using these libraries. **They do not produce deprecated libraries**: libqhull.so, qhull.dll, **libqhull_p.so**, and qhull_p.dll. Users of these libraries should convert their code to reentrant Qhull (libqhull_r.so) or link to libqhullstatic.a.

CarrotC, 2021-07-20:
hello, I get the same problem. Have you solved the BUG?

Pilat66, 2021-07-20:
I have the necessary library installed, but for some reason it is not available.

Workaround:

`LD_LIBRARY_PATH=/home/user/miniconda3/envs/pclpy/pkgs/qhull-2019.1-hc9558a2_0/lib/ python smooth.py`

But it seems to me that pclpy does not work correctly with conda and similar errors will come out. We need a mantainer.

sloretz, 2021-09-08:
Seems like this is the same issue as #90

CarrotC, 2021-09-14:
> Seems like this is the same issue as #90

There is no solution to issue #90 either

jvnsg, 2021-11-05:
@yqmy0814 
this seams to work! safed a lot of trial and error!

Divelix, 2023-01-17:
In my env (python 3.8) only this worked:
1. `conda install -c conda-forge/label/gcc7 qhull`
2. `conda install -c conda-forge -c davidcaron pclpy`

marrblue, 2023-06-27:
> you can create a new environment and install qhull first `conda install qhull==2019.1` then install pclpy `conda install -c conda-forge -c davidcaron pclpy`

Thank you @yqmy0814 It works!
I just reinstall qhull with version 2019.1 in my old conda environment and  the problem is gone.
`conda install qhull==2019.1`