davidcaron/pclpy #102: Hello! UnavailableInvalidChannel

459737087, 2022-02-18:
When I ran this code
_

> conda install -c conda-forge -c davidcaron pclpy

_

UnavailableInvalidChannel: The channel is not accessible or is invalid.
  channel name: davidcaron
  channel url: https://mirrors.tuna.tsinghua.edu.cn/anaconda/davidcaron
  error code: 404

You will need to adjust your conda configuration to proceed.
Use `conda config --show channels` to view your configuration's current state,
and use `conda config --show-sources` to view config file locations.

zhangzihui247, 2022-04-01:
Hello, have you solved this problem?
I think I meet similar situation, I type "conda install -c conda-forge -c davidcaron pclpy" in command line, but the installation progress has been 0%.

micdist, 2023-01-29:
problem still present to this day

459737087, 2023-01-30:
> Hello, have you solved this problem? I think I meet similar situation, I type "conda install -c conda-forge -c davidcaron pclpy" in command line, but the installation progress has been 0%.

check the conda version, and I nearly forgot how to slove it.